atc1441/ATC_MiThermometer #186: Publish every 30 seconds in stead of 5 mintues

sybux, 2021-04-25:
Hi there,

I've flashed successfully my thermometer, I was able to change some settings as disable battery display, and normal send every 5 minutes, but in my ESPHome logs, I can see that all my sensor are flushing data every 30 seconds.

I've just made the turn of my device to reset this value to 5 minutes, but no change. Is there a way to be sure that settings is set ?

Thx for help

atc1441, 2021-04-25:
Hey. Please try out victors version of this firmware. There the setting is simpler. And it is much more evolved, it is linked in the readme.md

In general the devices do send out notifications all the time and only change on new advertising time

sybux, 2021-04-26:
Sorry for late answer.... it's working like a charm ! thx

closed 2021-04-26 by sybux