atc1441/ATC_MiThermometer #188: Feature request: python script to read advertising packet

mexon, 2021-05-31:
I was hoping to use this firmware to regularly log the temperature on a raspberry pi.  So far I've found that I can run "sudo hcitool lescan" in one process and "sudo hcidump --raw" in another process.  But this is really cumbersome to do from a cron job.

I thought I might write a small python script, but when I google I find a bewildering number of libraries and examples.  All of them are designed to do something else, none of them seem capable of just reading an advertising packet.  I've investigated bluepy, Adafruit Blinka bleio, beacontools, pygatt, and now I'm just really confused.

Doesn't have to be python of course.  I did try scan-thermometer.js, but it has a bunch of dependencies not available in raspberry pi os.  The HTML pages are obviously not practical in this application.  Any suggestions for the simplest way to just get the current temperature from a thermometer? Headless, portably, and fairly cleanly?

ktims, 2021-06-02:
I came across the same problem, and would also appreciate some simple example code somewhere. The key step for which clear instructions are missing from almost everything I found online is that once you find the descriptor, you need to enable notifies, or it will never produce notify events:

```
# `c` is the characteristic object
c.getDescriptors(0x2902)[0].write(b"\x01\x00", True)
```

It is not exactly a trivial example, but I posted my code to collect the sensor data and dump it to influx in a [gist](https://gist.github.com/ktims/31e12dc8747b41a5edc0112b03230007), hopefully it is some help.

pvvx, 2021-06-04:
https://github.com/pvvx/hcitooladv

KR4DIO, 2021-06-09:
I wrote a little script to do this. It might not be the most elegant, but it works: [atclisten.py](https://gist.github.com/KR4DIO/8497f6fa6e324e213aafe4d2fabb5c46). note: run as root/sudo

ToXIc-Dev, 2021-06-16:
> All useful suggestions, thanks. Here's what I went with - it does what I need:
> 
> #190

Thanks for doing this.

One issue I am having with it, is that it displays the temperature wrong, for example it'll display 6.0 when the temp is actually 30.0(°C) on the display.

The other values are correct.

Any idea?

ToXIc-Dev, 2021-06-16:
> All useful suggestions, thanks. Here's what I went with - it does what I need:
> 
> #190

@mexon Thanks for doing this.

One issue I am having with it, is that it displays the temperature wrong, for example it'll display 6.0 when the temp is actually 30.0(°C) on the display.

The other values are correct.

Any idea?

mexon, 2021-06-16:
For me I do very occasionally get bad readings:

`{"timestamp": "2021-06-16T14:00:11+02:00", "mac": "A4:C1:38:XX:XX:XX", "temperature": 22.8, "humidity": 49, "battery_percent": 86, "battery_volt": 0.336, "count": 173}`
`{"timestamp": "2021-06-16T14:05:03+02:00", "mac": "A4:C1:38:XX:XX:XX", "temperature": 4.9, "humidity": 0, "battery_percent": 85, "battery_volt": 0.335, "count": 178}`
`{"timestamp": "2021-06-16T14:05:04+02:00", "mac": "A4:C1:38:XX:XX:XX", "temperature": 22.8, "humidity": 49, "battery_percent": 85, "battery_volt": 0.335, "count": 178}`

I thought at first that this might be data that was corrupted "on the wire(less)".  But BT advertising packets are protected by a CRC, so that should be impossible.  Instead I'm assuming it must be some glitch in the firmware.

Perhaps you could dump the full raw data from `val`?

ToXIc-Dev, 2021-06-16:
I'm getting bad readings every single time for some reason.

Here is the dump: "1a18a4c1388f354501341d300a5a59"

mexon, 2021-06-16:
Well the raw data looks good then, reading 30.8.  So no idea why the python is misfiring 🤷‍♂️

mexon, 2021-06-19:
@ToXIc-Dev yeah, dumb mistake, sorry about that: https://github.com/atc1441/ATC_MiThermometer/pull/194

gene1wood, 2023-12-21:
This work by @mexon has been merged and is available at

https://github.com/atc1441/ATC_MiThermometer/tree/master/contrib/python