atc1441/ATC_MiThermometer #189: Usabelity for other devices possible ? - here XiaoMi MCCGQ02HL Door sensor

andiluma, 2021-06-10:
Hi , i hoped and tried to use your great software for a newly ordered device. In general it is visible , also sending BLE information to my IObroker ,

advertised service  fe95 : 30588b090b4c1637ecaae408  ; look similar to the original information from unpatches Thermo devices 

 but no way to connect from your SW to change the default firmware . Maybe there will be a way to added such devices ? In general smaller and cheaper that the door devices for zigbee 
[image]

Device type
[image]

atc1441, 2021-06-10:
Please see the kompatibel devices in the following link https://github.com/pvvx/ATC_MiThermometer

To make it compatible it is an own project for itself

lolax80, 2021-07-29:
I there a fix to activate it MCCGQ02HL

unrelentingtech, 2021-09-25:
You can activate this device using the universal version of the flasher page:

https://atc1441.github.io/Temp_universal_mi_activate.html

kuenkin, 2022-11-04:
Hello,
I have this sensor but it doesn't do the activation through the previous page, it stucks in "Activating now, please wait"
Any clue?

camnop, 2022-11-29:
Same issue like @kuenkin