atc1441/ATC_MiThermometer #192: LYWSD03MMC battery draining fast on Release 36 default settings

baskapteijn, 2021-06-14:
Hello,

I have 4 LYWSD03MMC sensors. All 4 are configured with ESPHOME. 1 has it's original firmware, the other 3 have the ATC custom firmware. I'm monitoring temperature/hygro/battery levels with home assistent.

After about 1 week of runtime, the 3 devices with ATC custom firmware (running in custom advertisement mode and default settings) are all down to 86% of battery. The original firmware is still at 100%.

Now I can imagine the default settings for the custom firmware are not the most battery efficient. What I would like to know is what the "original" firmware interval/settings are, so I can match this with the custom firmware to maintain the same battery life. Maybe this can be added to the (wiki) documentation?

Also, there is no explanation of differences between the "mii like advertisement" and "custom advertisement". What does this mean? Please document this as well.

Thank you!

atc1441, 2021-06-14:
Hey. That is a normal behaivior and about 80% is 100%. For my Firmware

The reason is that i made 3.1V as 100% so its easier to tell how much more voltage the battery has with only knowing that value. Normally the battery has 3 volt at 100%

The stock battery % is not good, it stays at 100% till the battery  is empty then its 10%

Overall you should try out the pvvx firmware that one is way more advanced

baskapteijn, 2021-06-14:
Hi @atc1441 , thanks for the quick response.

So the battery life indication should probably not drop as hard from know on, there is a clear difference in interpretation between stock and custom firmware. Got it.

With regards to the advertisement type, and what the stock interval is, is that something you can document?

I will take a look at the pvvx firmware soon, good tip thanks!

Thanks a lot so far.

pvvx, 2021-11-14:
Typical behavior of a new battery:
[image]
Correlation with temperature:
[image]