atc1441/ATC_MiThermometer #196: Possible to attach a temperature probe to the sensor?

Sukalo0290, 2021-06-21:
Firstly thank you all for the hard work of this project! Everything working beautifully.

I have a quick question in regards to the sensor itself, would it be possible to remove the SHTV3 sensor and solder on a temperature probe instead to be able to use this sensor to detect the temperature of a specific point rather than air around the sensor . An example could be a waterproof sensor for an aquarium for instance.

Thanks for any insight / advice in advance into this.

atc1441, 2021-06-22:
Hey.
Yes that is possible, the stock sensor is not easy to solder, but if you are able to there is no problem.

Other people modified the device with a different Sensor then the code needs to be adapted of course.
Then you could already use a waterproof Sensor

Sukalo0290, 2021-06-22:
That's great news, would this be the probe of choice? 
Waterproof DS18B20-compatible Temperature Sensor
[image]

In regards to code, what code would need to be adapted? The python script or the firmware? Does anyone have any documentation / how to?

atc1441, 2021-06-22:
That sensor should work.

You can search for arduino code that uses it and adapt it to the c code of this repo.

You could adapt this code then https://github.com/atc1441/ATC_MiThermometer/blob/master/ATC_Thermometer/sensor.c

Sukalo0290, 2021-06-22:
Would anyone know the wiring diagram to the sensor too?

Red connects to 3-5V
Blue/Black connects to ground
Yellow/White is data

[image]