atc1441/ATC_MiThermometer #197: Possible to turn of display?

alexw1982, 2021-06-23:
Me (and I assume a lot of other people) utilize the unit mainly as a sensor with little use for the display in many cases. Would it be technically possible to turn of the screen to increase battery life? this would be awesome....

atc1441, 2021-06-23:
Hey,

the battery life does not increase by turning it of. 

The main power draw is the sensor reading.

i can not suggest on doing it

alexw1982, 2021-06-23:
Ok - thanks anyway :-)

closed 2021-06-23 by alexw1982

dgoadby, 2021-06-23:
If you really want the display off then you could hack the PCB and cut the VCC to the display. The LCD draw is minimal and, if you did kill the LCD power supply, you would lose the only obvious indication that it is working.

alexw1982, 2021-06-24:
> 
> 
> If you really want the display off then you could hack the PCB and cut the VCC to the display. The LCD draw is minimal and, if you did kill the LCD power supply, you would lose the only obvious indication that it is working.

Fully agree - it would have only made sense if there was a significant power saving benefit...