atc1441/ATC_MiThermometer #200: Black screen after flashing newest version

p4cx, 2021-07-01:
Also at first a big thank you for this amazing project! 
I updated three of my four LYWSD03MMC devices via the flasher on the newest firmware image (MHO_C401_v32n.bin) and now these devices seems broken. The screen won't show anything and also I can't find them via Bluetooth. I also tried (multiple times) removing the battery, but no luck at all. Do you think it is possible, that these are now waste :( 
Or is there something I should try to wake these devices up? 
Additional information: For flashing I used Chrome on macOS (I already had an older versions of the firmware on these devices, this worked really good - I only wanted to update these).

atc1441, 2021-07-01:
If the devices are really bricked you can repair them via a USB to UART converter

p4cx, 2021-07-02:
Thank you for your quick reply. At the weekend I will try to flash the firmware via UART in peace. Wish me luck. You are welcome to close this issue, I will definitely report here whether it worked.

atc1441, 2021-07-02:
Feel free to close it when it worked :)

p4cx, 2021-07-02:
So, it worked - all four sensors are working and I can reach them via the web ui by pvvx (I flashed his firmware on my devices, works like a charm!). Thank you again for this amazing project and your help.

closed 2021-07-02 by p4cx