atc1441/ATC_MiThermometer #204: Change the temperature offset

mrgfx, 2021-07-12:
Hey, my intention was to simply set the temperature off set and use the device with original mi home app...
I searched on the internet for the solution and the only solution I've found is to use this custom firmware. I've installed it and changed the temperature successfully. Unfortunately, I wasn't able to connect it with mi home app. Am I missing something with this custom firmware? Is there any method to simply change the offset and use it with xiaomi app? 
Thanks a lot! You are doing amazing job!!

atc1441, 2021-07-12:
Hey. 

Xiaomi app is not compatible with these custom firmwares and will not be sorry.

mrgfx, 2021-07-12:
Ok, sure. Do you know any other way to achieve it? I guess that it would be possible to clone the original code with uart to usb converter and modify it to show the temperature with correct offset. Do you think that it will be possible?

atc1441, 2021-07-12:
It should be possible yes. 

But not simple as it needs to be ifentified where the offset is set in the stock firmware

mrgfx, 2021-07-12:
Ok. Thanks. I'll post some updates here, if I'm able to change it or not, because somebody may find it useful