atc1441/ATC_MiThermometer #205: Can one use the Xiaomi home app after updating?

orenatan, 2021-07-15:
Hi
Can one use the Xiaomi home app to connect to those devices after updating to the ATC firmware?

atc1441, 2021-07-15:
Hey.

No that is not possible

orenatan, 2021-07-15:
Thanks for quick reponse!

Is there any advantages for your firmware if I don't use Linux anymore ?

Since I don't want to have an extra computer with Home assistant just for my smart home anymore, I moved to use Smartthings. So I wonder if there is a way to use Mijia BLE thermometers with SmartThings or to go back to the normal way with Xiaomi home app..

atc1441, 2021-07-15:
Any custom solution that can read Xiaomi Thermometers is compatible to the custom firmware, as there is the "MiLike" mode.

I dont know if SmartThings can do it.

But you can of course always go back to stock