atc1441/ATC_MiThermometer #207: Can't connect from mijia app even when flashed back to stock rom.

Blue-Beaker, 2021-07-19:
Can't connect from mijia app when flashed atc then flashed back to stock rom:
https://github.com/custom-components/sensor.mitemp_bt/files/4022697/d4135e135443ba86e403ecb2af2bf0af_upd_miaomiaoce.sensor_ht.t2.zip

Standard authentation failed(-29)
[image]

ahmadnazirafiq, 2021-08-18:
Same issue here. Flashed custom firmware successfully, reverted back to stock firmware. but Mi Home app can't bind the sensor anymore.

[image]

atc1441, 2021-08-18:
Please flash my firmware again once and then back to stock. This will erase the custom blocked did

ahmadnazirafiq, 2021-08-18:
I did, issue still persists. I managed to revert back to stock firmware from pvvx (https://github.com/pvvx/ATC_MiThermometer#bluetooth-advertising-formats) and this one works. Probably his stock firmware is older than yours.

atc1441, 2021-08-18:
Good to know, looks like they are blocking more stuff these days :)