atc1441/ATC_MiThermometer #208: Custom firmware reports incorrect humidity

ervu, 2021-08-01:
After flashing the MI LYWSD03MMC with the REL 38 firmware the display shows the wrong humidity over time.
On average the value on the display is 5% lower that the actual value after a couple minutes.

This can be simulated:
-remove the battery
-replace the battery
-correct humidity is on the display
-wait 6 cycles between humidity and battery display rotation
-after cycle 6 the wrong humidity is displayed
-also the wrong value is passed to bluetooth and home assistant

tested with multiple sensors.

atc1441, 2021-08-01:
Hey,

the 5% is the "normal" offset from this Firmware,

the custom firmware does calculations on the sensor data which is not done by the custom firmware. it was the offset setting added to be able to make a custom offset if it is disturbing.

the custom firmware uses the sensor values directly from the sensor

ervu, 2021-08-01:
If i understand correctly, the custom firmware always displays the wrong value and needs always to be adjusted manually?
Is it possible to add code to the firmware to add an adjustment of 10% by default?

atc1441, 2021-08-01:
Correct.

You can set the offset via the flashing website

ervu, 2021-08-01:
Is it possible to add code to the firmware to add an adjustment of 10% by default?

atc1441, 2021-08-01:
Sure you could add such a function quite simple in the code :)

ervu, 2021-08-01:
Thanks for the great code an the help !!

ervu, 2021-08-01:
> 
> 
> Sure you could add such a function quite simple in the code :)

Is this issued on the list for the next firmware release?

atc1441, 2021-08-01:
I am not actively working on the firmware right now.

Also the offset is not a default needed value so i will leave it like it is, everyone can set its own offset via the flasher if needed

ervu, 2021-08-01:
i have tuned the MI's with the offset. Works great :)
Keep up the good work.....

ProTofik, 2021-10-10:
I'm a bit confused by this one. If original firmware adds 5% to sensor's reading, what is the real humidity value? What sensor reports, or sensor + 5%? Which one is the closest to the real world value?