atc1441/ATC_MiThermometer #210: LYWSD02)) NotFoundError: No Services matching UUID {uuid} found in Device.

jhojin7, 2021-08-09:
Hi! I'm trying to flash your custom ROM to my LYWSD02MMC model.
Your web flashing tool wouldn't connect to my device with the log below. 

By the way I'm on firmware 1.1.2_0092,
and tried on both Win10 Chrome and  Android11 Chrome if that matters at all...

This is my log. `{uuid}` is my uuid. I masked it.

```
Log:
18:59:53: Searching for devices
19:00:00: Connecting to: LYWSD02
19:00:05: NotFoundError: No Services matching UUID {uuid} found in Device.
19:00:05: Reconnect 1 from 5
19:00:05: NotFoundError: No Services matching UUID {uuid} found in Device.
19:00:05: Reconnect 2 from 5
19:00:05: NotFoundError: No Services matching UUID {uuid} found in Device.
19:00:05: Reconnect 3 from 5
19:00:05: NotFoundError: No Services matching UUID {uuid} found in Device.
19:00:05: Reconnect 4 from 5
19:00:05: NotFoundError: No Services matching UUID {uuid} found in Device.
19:00:05: Reconnect 5 from 5
19:00:05: NotFoundError: No Services matching UUID {uuid} found in Device.
19:00:05: Something went wrong, to many reconnect's
```

Any help is appreciated. Thank You!

martmaiste, 2021-08-10:
@jhojin7 Pulling the battery out and putting back in and rebooting the computer helped me to get past this error just yesterday when trying different ROM versions.

nikito7, 2021-09-09:
This is for square lywsd03mmc