atc1441/ATC_MiThermometer #211: Feature request: use the subscription mode

miksumin, 2021-08-11:
Is it possible to make it use use the subscription mode as it working in original firmware?

atc1441, 2021-08-11:
Can you explain the subscription mode ?

miksumin, 2021-08-11:
In subscription mode in your code you send a request to the sensor firmware to register notifications for the requested service UUID and characteristic UUID and then you are just receiving messages from it in notification callback procedure and it is not nessesary to make periodic scanning and waiting for the advertizing messages. Such algoritm used in original firmware. Example of using it you can find in ESP32 core BLE library - https://github.com/espressif/arduino-esp32/tree/master/libraries/BLE/examples/BLE_client

atc1441, 2021-08-11:
Ok now i know what you meant. 

This is already implemented :)

Buut, as in the stock firmware it us needed to be constantly connected to the device in order to receive the notifications. This does draw more battery juice.

miksumin, 2021-08-11:
Yes, the constant connection uses more sensor battery.
So I can use my code working with the original FW in subscription mode with your FW without modifications?

atc1441, 2021-08-11:
Yes that should work