atc1441/ATC_MiThermometer #212: Using webtool to get CGPR1 bindkey results in "Not connected"

Timmiej93, 2021-08-29:
I'm trying to get the bindkey for the CGPR1 motion detector. I can connect to it just fine, but when I click "do activation", it always says "Not connected", even if it says "Connected" right above that. I have tried using my PC and my phone, which both have flashed LYWSD03MMC thermometers without issue.
I checked, the CGPR1 does have a Telink TLSR82** MCU (TLSR8253 IIRC), so I guess something else is causing the issue? Any help would be greatly appreciated.

unrelentingtech, 2021-09-25:
Similar problem with the RTCGQ02LM "motion sensor 2", I managed to activate one of them using a silly modification to the script, but another identical one did not work:

https://github.com/atc1441/atc1441.github.io/issues/10#issuecomment-927138149

pvvx, 2021-11-06:
> I checked, the CGPR1 does have a Telink TLSR82** MCU (TLSR8253 IIRC), so I guess something else is causing the issue?

plastic lever - associated with the pairing button. To switch to this mode, press and hold the button for about three seconds until the blue LED blinks ?

jturn08, 2021-11-07:
I had the same problem of not being able to get bindkey using Telink Flasher, and was able to successfully get the bindkey using exact steps below.

1. Used `[Telink Flasher v3.3](https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html)`
2. Open the back cover of the sensor and press the blue plastic lever button for about 3 seconds until blue LED starts blinking. Release the blue pairing button.
3. In Telink Flasher v3.3 webpage, select "Connect" button and then select "Qingping Motion & Light" device
4. Select "Do activation" button.

After a few tries, I was able to get the token and bindkey.

Timmiej93, 2021-11-07:
I did eventually manage to get one of my two CGPR1's working, but it took way too many tries, probably over 20. Anyway, that's an issue with the actual product, not with the software.

closed 2021-11-07 by Timmiej93

pvvx, 2021-11-07:
What's the point of registering with mijia if CGPR1 transmits all data and events in advertising packages via the unprotected Qingping protocol?
https://pvvx.github.io/CGPR1/

Timmiej93, 2021-11-08:
So you'd only need the MAC address in order to listen to the messages send by the CGPR1?

pvvx, 2021-11-08:
> So you'd only need the MAC address in order to listen to the messages send by the CGPR1?

Requires a sniffer or BLE Advertising receiver.

pvvx, 2021-11-15:
> I had the same problem of not being able to get bindkey using Telink Flasher, and was able to successfully get the bindkey using exact steps below.
>
> 1. Used [Telink Flasher [...]

CGPR1 out of the box is in the original unencrypted Qingping BLE ad format. In this mode, light and battery level are transmitted every second. And all events are sent in unprotected Qingping format. OTA works without bindkeys.
Switching to Qingping BLE mode and decoupling from Mi-Home is carried out by pressing the button for a very long time until the LED stops blinking.

jturn08, 2022-01-03:
> CGPR1 out of the box is in the original unencrypted Qingping BLE ad format. In this mode, light and battery level are transmitted every second. And all events are sent in unprotected Qingping [...]

How do you configure the CGPR1 in esphome? If I leave out bindkey, then I get an error `bindkey is a required option for binary_sensor.xiaomi_cgpr1`. I also get an error if I include empty string as bindkey.

pvvx, 2022-01-03:
> How do you configure the CGPR1 in esphome without including a bindkey? If I leave out bindkey, then I get an error `bindkey is a required option for binary_sensor.xiaomi_cgpr1`. I also get an error if I include empty string as bindkey.

[CGPR1_Utils.html](https://pvvx.github.io/CGPR1/CGPR1_Utils.html)

https://pvvx.github.io/CGPR1/

jturn08, 2022-01-04:
Yes, I was able to get CGPR1 bindkey using your utils.

Is there a way to configure CGPR1 in esphome while CGPR1 is set to the original unencrypted Qingping BLE ad format? Or is getting bindkey the only way to use CGPR1 in esphome right now?

pvvx, 2022-01-04:
ESPHome does not support Qingping format.

almirus, 2023-05-05:
@pvvx hi! here you wrote https://github.com/atc1441/ATC_MiThermometer/issues/212#issuecomment-968629758
it's work, device switch to Qingping mode, but how to switch pairing mode simultaneously?
i want to add to HA via [qingping integration](https://www.home-assistant.io/integrations/qingping/), no one device was found, however, here https://pvvx.github.io/CGPR1/CGPR1_Utils.html I've get all indicators

pvvx, 2023-05-07:
Home Assistant 2023.5.2

Bluetooth integration
[image]
Qingping integration
[image]
[image]
Bluetooth integration does not work with all BLE variants.
Write to the authors of Bluetooth integration. Bluetooth integration does not accept BLE advertisements subject to Bluetooth SIG standards.
---

Passive BLE Monitor integration
[image]
[image]

pvvx, 2023-05-07:
Bluetooth integration and its associated APIs do not work correctly with "passive scanning".
[image]
And in active scanning mode, all BLE devices are constantly being requested, draining the batteries and Bluez is constantly writing to disk, killing the SSD (TBW).