atc1441/ATC_MiThermometer #213: Devices dated 2021 04 not reporting

hackroyd, 2021-09-01:
I have previously used these devices in Home Assistant and programmed an ESP32, via ESP Home.
However, I recently purchased some more, bearing the same description as the originals, but dated '2021 04' on the packaging. These, were detected during flashing the ESP32 but they do not report any data. I have included a previously working device, with the new ones, in the ESP32 flash and this does work thus leaving me to assume that there has been a hardware change or the firmware is different to that I previously used. Unfortunately, as I did the original flashing via another laptop in Spain, and I am not there now, I don't have the ATC_Thermometer.bin file, that I previously used, with me.

I did the flashing using Vivaldi browser ( with chrome://flags/#enable-experimental-web-platform-features  enabled) on my laptop running KDE (Kubuntu 20.04)

atc1441, 2021-09-01:
Hey  please try out @pvvx version of the firmware ans see if it works better

hackroyd, 2021-09-01:
Thank you for your reply. There were 2 versions from that source. I tried both but, sadly, neither worked any better. Maybe there has been a hardware change :-(