atc1441/ATC_MiThermometer #220: Bricked 202105 HW

marcobe, 2021-10-05:
Just received 6 with motherboard date 202105, HW B1.9

Installed version 42 on it, and resulted in a bricked device after reboot.
Installed a version from November 2020 on it, same result.

Did the hardware change on these things or am I doing something wrong? Any suggestions/ideas?

marcobe, 2021-10-10:
Thank you! Got them both back working, so now wait for a fix? Anything I can do to help out?

SurfingSpider, 2021-10-11:
Following as I have bricked a couple of these too.

Korte68, 2021-10-16:
I bricked as well mine...now it is not connecting over BT...How can I repair it? Thank you

marcobe, 2021-10-16:
It looks like they are getting close to upgrading so then it will work, you can return it to stock, or just wait a week or or so ( me hopeful :) ).

Stock file is in this thread: https://github.com/pvvx/ATC_MiThermometer/issues/125#issuecomment-937552241

Korte68, 2021-10-16:
I was able to restore it...Thank you

gacek4000, 2021-12-26:
Hi, i uploaded ATC_Thermometer firmware via bluetooth. It worked but after pulled off battery sensor didn't wake up. I tried uploaded original firmware via UART with success but sensor still is blank :( 

[image]