atc1441/ATC_MiThermometer #228: Sensor reading interval

irarkh, 2021-11-02:
I was wondering why the sensor reading interval in custom firmware is very slower than original? And if there is a way to fix it back like original? The main issue comes up during setting generic thermostat and using ATC_MiThermometer as a target sensor. Specially in boiler and hydro heating systems which the temp difference is not fast, controlling the unit will be out calibrated due to sensor reading delay. I red some post Aaron said that sensor reading depends on temp difference.

Protoncek, 2021-11-02:
Slower? My experience is quite opposite: in original FW it can take up to 20 minutes from reset for temp and humidity to show up in HA, while in this custom FW data show up in a matter of seconds. Do you have set custom advertising? Don't use Mi-like one.

irarkh, 2021-11-03:
I guess you misunderstand me. I didn't mean advertising interval. Temp and humidity sensor reading itself. I use custom advertising and there is no issue at all. Everything works like a charm. But my problem is with sensing the input differences interval by sensors. In original FW is faster and reflection of sensors reading are 1s or less. But in custom FW is not the same.

Protoncek, 2021-11-03:
@irarkh : correct, i understood wrong. Now i tested my sensor: temperature display changes at intervals around one minute or so. In original FW this was indeed almost instant.
I already wanted to ask somewhere: what’s the difference between atc andpvvx firmware, and, which one to choose?

atc1441, 2021-11-03:
The slower update in my firmware is for power saving reasons. Readin the sensor takes the most power.

In a discussion long time ago it was agreed to make the update slower as it is very rare that you need Instant Updates

irarkh, 2021-11-03:
Thank you all participating in discussion. Aaron is there any way to make it optional on the next release? Or I need to recompile the changed #69 line myself?

irarkh, 2021-11-03:
> @irarkh : correct, i understood wrong. Now i tested my sensor: temperature display changes at intervals around one minute or so. In original FW this was indeed almost instant. I already wanted to ask somewhere: what’s the difference between atc andpvvx firmware, and, which one to choose?

To my understanding Victor's FW has a small difference in power consumption as well as some extra settings. Personally, I found Aaron's FW simpler and more user friendly.

sezlony, 2022-04-26:
hi!
I just flashed the ATC firmware onto 3 devices and I can confirm that updating interval in HA has been significantly shortened. With xiaomi fw update was inbetween 5-10 mins, now it is exactly 1 min.
I did not change any config after flashing, so I think this is default setting, but won't it affect negatively battery life?

I didn't like the xiaomi's original update frequency either, but I assume this drains battery too fast.

Any advice?