atc1441/ATC_MiThermometer #229: LCD Screen after Changing instand off

DJ3vil, 2021-11-06:
Hey, ive flashed a year ago 3 with great succus.
Now ive moved to a new home and needed 2 more, so ive buyed the same one again. 
After Flashing the latest Firmware it restarted and worked. After changing the Advertising Type to Mi Like and the Smiley it went off from now on, regardless what im doing. I can still connect to the Board

Andy help ? 
Thanks

atc1441, 2021-11-06:
You tried replacing the battery ?

DJ3vil, 2021-11-06:
Yea tryed it. The Second that came workes great

atc1441, 2021-11-06:
Hard to gues.

You can try to reflash via a USB to UART converter as descriped in the repo

DJ3vil, 2021-11-06:
Unfor. i dont have eqip to open it. Is there a way i can flash back the original software ? So i can look, the LCD may be damaged ?

Bulls91, 2021-11-11:
i have the same problem. I buyed three new ones and after flashing the screen stays off. Did anyone have an idea how that problem can be fixed?

atc1441, 2021-11-11:
Please download the latest firmware version. 

There where a few updates lately to new hardware revisions

atc1441, 2021-11-11:
https://youtu.be/orinnMN1Ik4

DJ3vil, 2021-11-13:
Same 4 me, ive already used the new one

foxsam21, 2021-12-07:
I just flashed the latest version `70` and I am having a similar issue. The lcd stops updating after a while. I took out the battery and replaced it and the screen went blank. From the flasher page I was able to reset the settings and the screen woke up for a while but then just stuck on one temp again. In HA it is reading the right temp so the sensor is working its just the display.

JohnMoor, 2021-12-12:
I have the same. 

1. I flashed the custom firmware (latest ATC_thermometer.bin file from 28Oct 25fa534) using the atc1441.github.io/TelinkFlaher.
2.  All went well. 
3. Replaced the battery and after reinserting the screen stays blank. 

Device is detected and custom firmware loaded also able to change settings, but nothing shows on screen. 

Tried:
- Replaced with other new battery the same. 
- Even re-flashed the device with the (same) custom firmware.
- Left out battery for 5 min

But non of the above helped.

JohnMoor, 2021-12-12:
Just to confirm, I tried to replicate this problem and indeed after removing the battery again after flashing custom firmware display was blank again. Restoring stock firmware fixed the problem.
I tried it a 3rd time but now, clicking "Save current settings in Flash" after making some config changes. (I didn't do this the 2 times before). Now the display keeps working after reinserting the battery.

madcrokodile, 2021-12-28:
Hi! I have 4 sensors, and get the same issue on 2 of them. Maybe this info helps to understand situation.

If I connect sensor to mi app before any flashing movements - things goes ok.
If I'm not connect sensor to mi app and activate/flash it directly from webapp - I get LSD lost issue when battery replacement.

Also, sensor with lost LSD still works, it can be discovered from homeassistant plugins for example.
Only display not working.

sahomm, 2022-01-19:
I has had the same problem. You can solve that to be flash the stock Fw and then the custom FW. I now that is a bad solution, but it works ;)