atc1441/ATC_MiThermometer #230: Extend flash.py to do the activation part

Juweii, 2021-11-08:
Hello,

is it possible to extend the flash.py script to include the activation part?

````
# python3 flash.py -f ATC_Thermometer.bin A4:C1:38:xx:xx:xx
Trying to connect to A4:C1:38:xx:xx:xx
Failed to connect to peripheral A4:C1:38:xx:xx:xx, addr type: public, retrying: 1/5
Connected to A4:C1:38:xx:xx:xx
Detected Mi Thermometer. Can't flash it.
````

After it detected the Mi Thermometer with stock firmware it should be able to activate it and trying to flash again?
This would make this script an easy way to flash new devices and all-inclusive tool instead of doing it with the browser based tool.

What do you think?

Best regards,
Juwei

atc1441, 2021-11-08:
Hey. 
There was  discussion about it some time ago already. 
The Activation part involes the Encyrption handschacke and it was enough work to it it working in the web flasher so i will not add this to python.

Someone who wants to can of course port the activation :)

Juweii, 2021-11-08:
Ok, I see.
Thank you for the quick response! :)

closed 2021-11-08 by Juweii