atc1441/ATC_MiThermometer #231: Flashing another FW after activation and using custom firmware on LYWSD03MMC

srdjansimic, 2021-11-14:
Hello,

Following instructions, I flashed custom FW, release 70. Now I wanted to do some changes and when I try to connect I get the following:

11:24:19 AM: Detected Mi Thermometer
11:24:19 AM: NotSupportedError: GATT operation failed for unknown reason.

I repeated this on another unit and it is the same? What am I missing? What is the correct procedure to reflash unit that has custom firmware?

Best regards
Srdjan

Protoncek, 2021-11-14:
Such errors happen a lot in my case. I just repeat connection a few times and once it connects…

atc1441, 2021-11-14:
It could be a nearly empty battery.

please see if the device reboots after connection

srdjansimic, 2021-11-14:
Thank you for prompt answer. No, the battery is ok, this is a new device. Battery is at 100% and it does not reboot. It just tries 5 times and gives up. I finally tried another phone with another Android release and it worked. So we will assign this problem to my old android version :) But is was strange for me because it worked ok with stock firmware. 

So, the reason I was doing this is when this unit was working with stock FW and a key, my raspberry pi was plotting temperature correctly. Now with this new release, is seams like the unit is out of the reach. When I bring it closer all is fine. Is there a power setting for bluetooth that can be set?

Once again thank you for your prompt answer, I thought I was doing something wrong.