atc1441/ATC_MiThermometer #236: Error loading sensor at Homebridge startup due to "same UUID"

paig0n, 2021-12-07:
Hello!

After connecting sensor with original FW, using Bind Key value for decrypting I receive the following at Homebridge startup:

> `[07/12/2021, 17:09:01] [Temperature & Humidity] Error loading the accessory "Hygrotermograph" from "homebridge-mi-hygrothermograph" requested in your config.json: Cannot add a bridged Accessory with the same UUID as another bridged Accessory: f6cf49dd-9a64-4313-82fd-1c0156a6866f
> `

I have no other sensor connected but before I had the same sensor connected with the updated FW. So at this point I reflashed the original FW on the sensor and connected it using Bind Key. Is it possible that the UUID got stored somewhere where I can't see it? 
Also for clearing other doubts I reflashed the original SW because the values were not updating when inside Homebridge and the values where anyhow seen only inside the web interface but never shown in Home app. 

Let me know if you have any doubts or queries in finding the root cause. 

Thanks!
Roberto

WhyElEss, 2021-12-12:
Did you try to "remove single cached accessory" the Homebridge settings?

paig0n, 2021-12-12:
> Did you try to "remove single cached accessory" the Homebridge settings?

It does not appear under cached accessories in settings so i think it got automatically removed from there