atc1441/ATC_MiThermometer #237: Display off after Battery Change

DerChrisBo, 2021-12-16:
Hey guys,

I just flashed 4 of my Sensors and I'm pretty happy about the result. 

I just have the issue, after changing the battery the screens keeps blank, but continues to send data to HOmeAssistant. For me it's not a big issue, because I can view everything in the app, but standing in front of the sensor it would be pretty cool to see the screen...

What did I do wrong?

Thanks in advance!

atc1441, 2021-12-16:
Hey, sounds like you got one of the B1.9 version sensors, unfortunately that is not fully reversed by me as i dont own one.
https://youtu.be/orinnMN1Ik4

You can try to use pvvx version of this firmware and it should work fine.

Greetings

DerChrisBo, 2021-12-21:
That solved my issue!
Thank you very much!

closed 2021-12-21 by DerChrisBo