atc1441/ATC_MiThermometer #239: Complete death

muckie89, 2022-01-08:
Flashing latest firmware on a hw 1.9.
No Bluetooth working, no screen working. 

Changed Battery still death

atc1441, 2022-01-08:
Hey. 

Sounds like you need to use the usb to uart method to revocer it.

Greetings

sahomm, 2022-01-19:
ohhh no! I have the the same observation, but the device wasn't death! 

I've flash the current custom FW all was fine. About another issue I've released the battery an now the display are totally off.
After I have refresh the stock FW the display are where ok. When flash the FW and release the battery for more than 10 sec, then the is the display off and it looks like the device are death. 

@atc1441  should I open about that a new issue?