atc1441/ATC_MiThermometer #242: please explain more on temperature offset

supercazzola, 2022-01-23:
I understand that the first two hex vales to send for correcting  temperature offset are FA and then the next two should be the offset where you need to convert an int8 to hex.
I just do not know how to convert an int8 to hex. I looked online and can not find a table.
I see that the command FA00 is sent when you tell the firmware webpage to reset to zero offset. I assume FA01 would change the offset by +1 degree C. I would like to know how to represent negative numbers for example, how do I tell the device I need to have the offset be 2 degrees C cooler than what it displays right now?

OscarHanzely, 2022-01-28:
You first get the value you want. Let's say +2.0C
that's represented int8 = 20
you convert it to hex = 14
you type in FE14 and click Send

Alllfff, 2022-02-28:
what about a negative offset? thanks

robertely, 2022-09-21:
Hate to bump a question, but is there a way to set a negative offset ? All of my sensors seem to read high for what ever reason

LawDeveloper, 2022-10-25:
Same here, I really would like to set a negative offset. Anyone nows how?

jalbarl, 2022-11-11:
Has anyone tried sending FEEC? It should result in an offset of either -2.0 or +23.6. I don't have a device at hand to test it.