atc1441/ATC_MiThermometer #244: Release 75 : Impossible to connect to the device

Sardine29, 2022-01-29:
Hello,
I have installed the firmware 75 on the device . No issue . 
The device seems well updated . I have new Mac adress, I can see it on Bluetooth but I cannot connect to it . Or only 1 or 2 second and it is disconnected immedialtly 

I have this log 
Log:
15:00:05: Searching for devices
15:00:08: Connecting to: ATC_F2FA4A
15:00:11: Detected custom Firmware
15:00:16: Disconnected.
15:00:28: Searching for devices
15:00:53: NotFoundError: User cancelled the requestDevice() chooser.
15:00:53: Reconnect 1 from 5
15:01:04: Disconnected.
15:01:04: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:01:04: Reconnect 2 from 5
15:01:12: NetworkError: Connection failed for unknown reason.
15:01:12: Reconnect 3 from 5
15:01:26: Disconnected.
15:01:26: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:01:26: Reconnect 4 from 5
15:01:34: Detected custom Firmware
15:01:54: Disconnected.
15:02:06: Reconnect
15:02:28: Disconnected.
15:02:28: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:02:28: Reconnect 1 from 5
15:02:28: Disconnected.
15:02:28: NetworkError: GATT Server is disconnected. Cannot retrieve services. (Re)connect first with `device.gatt.connect`.
15:02:28: Reconnect 2 from 5
15:02:35: Detected custom Firmware

What could be wrong  ?
I cannot connect to the device so I cannot download a previous firmware.

Thanks in advance for you help

Rgds

aames, 2022-02-08:
yeah I have this too, did it brick yours?

atc1441, 2022-03-01:
You can always recover via UART.

What hardware revision is it? 

If 2.0 please try out victors firmware version

the-mrd, 2022-07-06:
Can confirm, release 75 stuck at "Detected custom Firmware". Tried removing battery, doesn't help.

m4dm4xi, 2022-08-19:
Same for me.

Cant connect anymore

ars3nicum, 2022-08-29:
Had the same issue. Stuck on "Detected custom firmware", but I tried to select stock firmware file and Flash and it worked,  stock firmware successfully flashed!