atc1441/ATC_MiThermometer #245: ATC_Thermometer.bin is detected as alternative firmware by TelinkMiFlasher

varsanyim, 2022-02-01:
After flashing the LYWSD03MMC with the ATC_Thermometer.bin from the Git I recieved strange errors on Telink Flasher (see below). 

10:38:07: Connected.
10:38:15: File was selected, size: 64320 bytes
10:38:15: Count: 4020
10:38:17: Start DFU
10:53:13: Update done after 894.735 seconds
10:53:18: Disconnected.

https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

11:31:14: Connecting to: ATC_3CF95A
11:31:23: Detected custom Firmware
11:31:24: Detected alternative Firmware? Use: https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html

atc1441, 2022-02-01:
Seems like you are mixing my ( atc1441 ) and victors firmware

varsanyim, 2022-02-01:
Oh, damn, I see the user error! I should have used https://atc1441.github.io/TelinkFlasher.html. 

I reflashed with your link and you firmware.

"Select Firmware: ATC_Thermometer.bin
Status: Update done after 38.584 seconds"

11:58:03: File was selected, size: 64320 bytes
11:58:03: Count: 4020
11:58:09: Start DFU
11:58:48: Update done after 38.584 seconds
11:58:52: Disconnected.
11:59:14: Reconnect
11:59:20: Detected custom Firmware