atc1441/ATC_MiThermometer #249: Advertising interval not working?

JGeeforce, 2022-03-07:
This weekend I flashed 2 sensors, this went very smooth. Thanks for the tools and information. I want to use them to get rid of my old 433MHz temp/humidity sensor. 

I installed an ESP32 and left it running this weekend in Home Assistant. All seems to work fine, but:

1. I configured the sensors with advertising intervals of 1 and the other 5 minutes. But when I look in HA at the last changed and last updated information, I see 1 hour/1hour for example. For the battery of one of them, I even see 9 hours ago. I would expect that at least the last updated part should be 1min or 5min. To compare this, the 433MHz sensor updates every 60 seconds, and that is indeed what I see in HA. (I clicked the 1 and 5 min button, looked at the logs and saved the config. LCD showed STO).

2. If I take out the battery from a Xiaomi device, this is never seen in HA. After 2 hours, the status in HA is last seen/last changed 2 hours.

Protoncek, 2022-03-07:
What do you run in esp32? Esphome,  tasmota or any other fw?

Protoncek, 2022-03-07:
Switch to tasmota. I’ve had esphome with 8 sensors and interval was pretty random. After esp32 reboot it could take up to 5 minutes (sometimes even more) to get first reading in HA, while with tasmota all readings come in the rane of seconds after restart and are in fact published every 60 seconds (or more frequent if you change interval). Supposely esphome (basically arduino, if i’m correct)  has very buggy BTdrivers.
Just install tasmota32 for bluetooth, set up MQTT on HA and you will be happy user…

JGeeforce, 2022-03-07:
Hmm ok. I need a little help then! Just googled a bit, have to download a tosmata32 bin file, but how to flash the ESP board? And is the programming the same after that?

Protoncek, 2022-03-07:
Yes, choose tasmota32 bluetooth. You can update via esphome’s website (enable website and update via OTA). After reboot you’ll have to connect to tasmota’s wifi  (192.168.4.1] and set your wifi data. After you get into tasmota’s webpage just enable BT and tasmota will discover your xiaomi. 
then in HA install tasmota integration and set up mqtt sensors.