atc1441/ATC_MiThermometer #251: Battery consumption/drain

JGeeforce, 2022-03-13:
I installed your firmware on the 8th of March, also because it could function longer on a battery. Some days before that I installed 2 others with the other 1441 firmware. I use default settings and custom advertising. It is working fine, but the battery went from 100% (brand new out of the box) to 77%. With that speed it will take 4 weeks to replace batteries. 

With the 1441, which are running several days longer, are on 84% at the moment. So also fast to my opinion, but one of them is set to advertise every 1 minute.

Why does it use so much battery? Do I need to adjust some settings? Or are the batteries supplied by Xiaomi bad ones?

atc1441, 2022-03-13:
Hey.

This is because ~80% shown in display is ~100% in reality. 

I made 100% to be higher then 3V because otherwise it would cap at 3V and 100% and you would not know how much higher the actual voltage is.

The batterys claim to be 3V at 100% but in reality are a out 3.2V at 100%

In conclusion the percent will now stays around where it is for a long time.
Also the OTA update does pull the voltage a bit down and will even recover over time

JGeeforce, 2022-03-13:
Hey, Great, thanks for the explanation. And your work!

tobiasdroessler, 2022-08-06:
Are there experiences (or calculations)  how long a new battery lasts?  Assuming default settings and no special BLE activity.

traeu, 2022-08-17:
This is the battery percentage I recorded from 4 LYWSD03MMC.
After nearly one year, they are still working with the first set of batteries.
But you need to use good batteries. Some of my Xiaomi thermometers came with batteries which where empty after two months.

[image]

GalvaniObst, 2022-08-25:
@traeu what batteries do you use and to what did you set the advertise interval?

traeu, 2022-08-25:
@GalvaniObst Advertise interval is set to default 2500ms. The batteries in the units on my screenshot are the original ones that came with the thermometers. Something is written in chinese on them, I can't tell you the brand.
I don't have a recommendation for good batteries...I just know based on different purchases that they can come with good and bad ones. So don't blame the custom firmware too quickly if the battery dies fast.

GalvaniObst, 2022-08-26:
My LYWSD03MMC weren't delivered with batteries and I'm using not a brand product.
I didn't want to blame the custom firmware. Initially I've set the interval to 10s and wasn't surprised that this consumes too much energy. With 1m it seems acceptable but I guess I will increase it to 5m.

sezlony, 2022-11-29:
we might need a bit of fine-tuning, because it's impossible that battery is shown depleted since 5 days and sensor still operates
[image]