atc1441/ATC_MiThermometer #252: how to change model_id when flash custom LYWSD03MMC firmware

0neday, 2022-03-19:
[image]

HA will get temperature from ATC_53B8C8 for ATC_63BB0A sometimes, when  ATC_63BB0A  is down.  I found that model_id is same both ATC_63BB0A  and ATC_53B8C8 

so, how change model_id ?

michapr, 2022-03-19:
Why you want to change the model_id?
It describe the device type - not the device "number" or specific device.

The "name" is the right selector for the device value.

Protoncek, 2022-03-19:
OF course model id is the same, it's same kind of device, correct? You have sensor setup wrong in HA if readings are messed up. Correct identifier is the one with MAC addres inside, in your case ATC_53B8C8 and ATC_63BB0A (identified "name").

0neday, 2022-03-19:
thank you, maybe produce entries by OMG automatically have some problem. I will add sensor by manual integration

closed 2022-03-19 by 0neday