atc1441/ATC_MiThermometer #255: Battery gets to 35% then dies quickly

callacomp, 2022-04-16:
I've around 8 devices and they all seem have the same characteristics of dying at around 35% battery level. Is there a firmware update or us this an issue.
[image]
[image]
[image]
[image]

pvvx, 2022-07-10:
This is a hardware problem in Xiaomi LYWSD03MMC. There is no capacitor soldered to the power rail on the board. Savings from Xiaomi.
[image]

callacomp, 2022-07-11:
Ok thanks, any idea what cap they are, there's room for a small one behind the back case? Mind you, just taken one apart and the screen fell off, can't getting to display again but still working and reporting.

Metaln00b, 2022-08-02:
That would be interesting to know, but there is no marking on the pcb, in which direction the tantalum-capacitor has to be soldered.

tcurdt, 2023-08-07:
> That would be interesting to know, but there is no marking on the pcb, in which direction the tantalum-capacitor has to be soldered.

Maybe worth trying both directions and sacrificing one. 🤔

pvvx, 2023-08-10:
After 2 years, the thermometer I used on the street deteriorated. This is version B1.9. The flux on the battery contacts and under the chips was not washed off in production. I checked the others - the situation is similar. The thermometer abruptly began to consume more than 2 mA (constantly, in the sleep mode of the microcontroller, instead of a pair of uA). Flushing the printed circuit board did not give anything. Only the heating of all components by the soldering station helped.

poseidogithub, 2023-11-11:
> After 2 years, the thermometer I used on the street deteriorated. This is version B1.9. The flux on the battery contacts and under the chips was not washed off in production. I checked the others - the situation is similar. The thermometer abruptly began to consume more than 2 mA (constantly, in the sleep mode of the microcontroller, instead of a pair of uA). Flushing the printed circuit board did not give anything. Only the heating of all components by the soldering station helped.
> 
> This ended the life of the second battery in a couple of days: [image]

sorry for the offtopic, but that's a nice graph :D from where is it?