atc1441/ATC_MiThermometer #256: Display no longer working after battery replacement

dasrecht, 2022-04-16:
Hi there, 

I've been running the custom firmware for a while and noticed that one of my devices stopped sending data, so I replaced the battery on it. Interestingly it didn't start to send data, but I was able to flash the newest firmware onto the device.

I could reproduce the display not returning to life on two other sensors. They all work sending data via Bluetooth, but the display stays off. Did anyone else run into similar issues with their Mi Thermometers?

I'm puzzled that the display wouldn't work, but the rest of the functionality is not suffering any issues.

ShadNex, 2022-05-06:
I’m experiencing the same on my devices (2 out of 3) after battery replacement. I don’t know if it has anything to do with the custom firmware though. Might be a hardware issue?

dasrecht, 2022-05-21:
Whoo confirmed flashing the stock firmware brought back the display.

dasrecht, 2022-05-21:
I was able to reproduce the issue with the latest firmware - if the battery powercycles the display stays off. but connectivity still works.
If the battery is changed quickly (within 2-3 seconds) the display comes back to life.

OniLink57, 2022-06-19:
Got the same problem today. The screen went off and stayed off after the battery change. I had to flash the stock firmware then the latest version. Thank you all for your feedback which helped me :)

pvvx, 2022-07-02:
For version B1.9 in the ATC firmware, the I2C frequency is too high for the display controller.
If the temperature or battery voltage is not normal, then the display will fail. Basically, this manifests itself as a lack of readings on the display and re-initialization does not help.
Remove and insert the battery for a long time. This may help in the short term. But in the software it is required to reduce the speed of the CLK I2C...

atc1441, 2022-07-02:
Just updated the repo to your speed victor, hope that helps

pvvx, 2022-07-02:
https://github.com/pvvx/ATC_MiThermometer/blob/master/src/lcd.c#L147

luizrrocha, 2022-07-05:
Tried with Release 78, no change. Had to install the original firmware and then flash release 78. After quick battery change, display did not return as before.

marxus4, 2022-08-12:
I have the same on 3 sensors

arepih, 2022-08-28:
Same here. I have about 30 sensors. Flashed with release 78, all of them lost display after battery change.

nguyenmanhthao996tn, 2023-02-09:
Hi, this issue isn't solved yet, right? I had the same problem with my devices (B1.9 & Release 78) this week.

mslio, 2023-09-10:
Hi, I must confirm, it is not solved yet. I have used release 76 instead.

the78mole, 2023-10-16:
I had a sensor that had an empty display. I took out the battery, placed it again, but nothing happended. I took it out over night and today it showed wired behaviour. I put in a new battery and the display began to get crazy. The LCD segments were dancing around. It was not a random sequence, instead it repeated somehow. I then reflashed the firmware and it was dancing to another "song". I again reflashed the firmware with the 4.4 Beta and now, it shows a static screen, but with no useful information. BT still works. I can connect, it sends out updates with BThome, but the screen still shows the same segments.

mslio, 2023-10-19:
The display stopped working for me after uploading the last custom release. Uploading the original fw only helped. Then I uploaded a custom release one older and it still works.