atc1441/ATC_MiThermometer #257: going back to stock FW takes forever

sezlony, 2022-04-27:
I wanted to revert to stock firmware from ATC because battery draining issues, and it is taking a vast amount of time!

It's already doing it for 1020ish seconds and progress is still 77% only, so you have to be very careful with inital battery level not to run dry before it finishes flashing

did anyone experience the same?

Neotastisch, 2022-05-28:
How do you revert to the stock firmware anyway?

Cougar, 2022-09-10:
I'm not sure if the stock firmware is any way different. I updated around ten thermometers today from original and ATC to the latest ATC release and one of them (older ATC) took around 15 min to flash. I had to make sure that my phone didn't go to sleep (happened first time).

atc1441, 2022-09-11:
Hey, sometimes the TLSR micro does not change correctly to the fast BLE mode, if that happens you can just restart the whole connection and update and it should be fast again.

sezlony, 2022-09-11:
one does not simply do restart while uploading firmware

atc1441, 2022-09-11:
Its no problem to cancel an ongoing update on these devices.
No changes will be made until the final and last part of the firmware is transmitted :)

closed 2023-12-07 by sezlony