atc1441/ATC_MiThermometer #263: 1/4 sensors suddenly won't connect to ESPHome anymore

mortenmoulder, 2022-08-16:
A couple of weeks ago, I changed the CR2032 battery in one of my LYWSD03MMC sensors. Ever since then, I have not gotten reliable readouts. I just tried flashing brand new firmware today, for the first time in over a year. As you can see, ESPHome can easily find the device:

```
[17:12:33][D][esp32_ble_tracker:620]: Found device A4:C1:38:97:BA:BA RSSI=-62
[17:12:33][D][esp32_ble_tracker:641]:   Address Type: PUBLIC
[17:12:33][D][esp32_ble_tracker:643]:   Name: 'ATC_97BABA'
```

but nothing happens. 3/4 other sensors work perfectly. Here's how I set it up in ESPHome:

```
sensor:
  #ATC_97BABA
  - platform: xiaomi_lywsd03mmc
    mac_address: A4:C1:38:97:BA:BA
    bindkey: "9bcffe15REDACTED93f36449c1a32d"
    temperature:
      name: "Kitchen LYWSD03MMC Temperature"
    humidity:
      name: "Kitchen LYWSD03MMC Humidity"
    battery_level:
      name: "Kitchen LYWSD03MMC Battery Level"
```

Any clue why? I can easily connect to the device via my phone, flash new firmware via TelinkFlasher, and so on. It just won't connect to my ESP32, even though it's literally right next to it. I also tried yet another battery, but it still didn't change anything.

Protoncek, 2022-08-16:
If you flashed pvvx FW - why do you stull use original xiaomi platform? Use pvvx_thermometer, that way you don’t need bind key.

mortenmoulder, 2022-08-16:
@Protoncek Okay, I thought this firmware was still the way to go. I guess not. Thanks!

Protoncek, 2022-08-16:
You can also use atc_thermometer, sure (only now i saw that it’s atc github, not pvvx, sorry).