atc1441/ATC_MiThermometer #264: NotFoundError: No Services matching UUID {uuid} found in Device after activation and flash

barbergeek, 2022-08-19:
Hi. I'm trying to set up some new LYWSD03MMC thermometers. I'm using the current release 78 firmware. I could connect and activate the new devices, but after I flash it with the firmware, I only get the following error:
NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device

It tries 5 times and then gives the Something went wrong, too many disconnects error.

Is this bricked now? It won't connect sufficiently to flash something else.

I'm running on a pi on chromium (which was the only thing I could get to connect consistently... Windows wouldn't work) 

Thoughts?
Thanks!

atc1441, 2022-08-19:
Hey, please try out this version and see if it fixes it :) 
https://github.com/pvvx/ATC_MiThermometer

barbergeek, 2022-08-19:
Thanks! I was actually able to recover using the OTA version https://pvvx.github.io/ATC_MiThermometer/TelinkOTA.html

atc1441, 2022-08-19:
Perfect

PieBru, 2022-08-20:
Same error here.
I'm with Chromium on my Arch box.
Failing to connect I am not able to re-flash the device with pvvx FW.
Any hint?
Thank you for your great findings and work.
Piero

pvvx Telink flasher log:
```
11:36:36 AM: Searching for devices
11:36:42 AM: Connecting to: ATC_4646EE
11:36:55 AM: NetworkError: Connection failed for unknown reason.
11:36:55 AM: Reconnect 1 from 5
11:37:01 AM: NetworkError: Connection failed for unknown reason.
11:37:01 AM: Reconnect 2 from 5
11:37:11 AM: Not found Telink OTA service!
11:37:11 AM: Disconnected.
```

pvvx Telink OTA log:
```
11:38:07 AM: File was selected, size: 67252 bytes
11:38:07 AM: Count: 4204
11:38:16 AM: Searching for devices
11:38:20 AM: Connecting to: ATC_4646EE
11:38:27 AM: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
11:38:27 AM: Reconnect 1 from 5
11:38:27 AM: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
11:38:27 AM: Reconnect 2 from 5
11:38:27 AM: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
11:38:27 AM: Reconnect 3 from 5
11:38:27 AM: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
11:38:27 AM: Reconnect 4 from 5
11:38:27 AM: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
11:38:27 AM: Reconnect 5 from 5
11:38:27 AM: NotFoundError: No Services matching UUID 00010203-0405-0607-0809-0a0b0c0d1912 found in Device.
11:38:27 AM: Something went wrong, to many reconnect's
```

PieBru, 2022-08-20:
I had no luck connecting with an Android v10 tablet and a v11 phone.
I greatly prefer OTA instead of UART flashing.
Piero

analogic, 2023-01-05:
Ubuntu here

Flashing then restarting computer and then connecting again helps. I had some success also with removing device from "known devices", disabling&enabling bt and restarting chrome.

I've not tracked down right spot of the bug, but it seems that the problem lays in chrome or some linux layer (bluetoothd?) which is caching some part of device info (service list?).

holefrog, 2023-01-06:
> 

You are right, same using ubuntu. 
In bluetooth settings, remove paired ATC_Device, then connect again.