atc1441/ATC_MiThermometer #271: Flashed MiThermoter does not connect to Home Assistant

badsoden, 2022-12-04:
Hi, I flashed it with the lastest firmware from the project but it does not show up in home assistant. Will there be any solution for this?

stefsims, 2022-12-26:
I have the same issue. Flashed a LYWSD03MMC 1.4 with v3.8 and it is not showing in Home Assistant.

tasssman, 2023-02-13:
The same. Oryginal software works without problem, but on newest firmare is not discovered by HS.

stefsims, 2023-02-13:
Oh forgot about this... I changed the thermometer advertising type to BThome and Home Assistant found it instantly.

boonya, 2023-02-14:
Guys, as far us I remember by flashig with this custom firmware your device starts to interact differently. In other words it's API changed. So, first of all I would suggest you to investigate logs of the device which interacts with your thermometer.
In my case I use ESP32 as a bluetooth geteway, so in it's config I had to change `platform: xiaomi_lywsd03mmc` to be `platform: atc_mithermometer`.
https://esphome.io/components/sensor/xiaomi_ble.html#lywsd03mmc

geeneo, 2023-02-14:
> Oh forgot about this... I changed the thermometer advertising type to BThome and Home Assistant found it instantly.

Please can you clarify Advertising Type:  I  only see Custom and Mi Like in the options.

ajbanck, 2023-02-16:
'Mi Like' worked for me but requires the bind key to be configured in HA. For Custom the 'Passive BLE Monitor integration' integration needs to be installed from HACS

tasssman, 2023-02-17:
> 'Mi Like' worked for me but requires the bind key to be configured in HA. For Custom the 'Passive BLE Monitor integration' integration needs to be installed from HACS

Confirm. Worked with BTHome option (custom firmware) with Passive BLE Monitor integration.

stefsims, 2023-02-18:
No need for Passive BLE Monitor as it is now native in Home Assistant.

Misiu, 2023-03-27:
Will this work with ESPHome on ESP32C3?

```
esphome:
  name: esphome-web-25bf30
  friendly_name: BTProxy
  platformio_options:
    board_build.mcu: esp32c3
    board_build.variant: esp32c3  

esp32:
  board: esp32-c3-devkitm-1
  framework:
    type: esp-idf
    sdkconfig_options:
      CONFIG_BT_BLE_50_FEATURES_SUPPORTED: y
      CONFIG_BT_BLE_42_FEATURES_SUPPORTED: y
      CONFIG_ESP_TASK_WDT_TIMEOUT_S: "10"

# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "my-key"

ota:

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Esphome-Web-25Bf30"
    password: "my-pass"

esp32_ble_tracker:
  scan_parameters:
# Adjust timing if the defaults do not work in your environment
#    interval: 1100ms
#    window: 1100ms
    active: true

bluetooth_proxy:
  active: true
```

I've turned on bthome advertising, but nothing is found in HA.
What other integration I must use? Maybe my config is wrong?
I run Home Assistant on a Dell Wyze terminal, so I don't have Bluetooth on that machine, only ESPHome Bluetooth proxy.
I have MHO-C401 and CGG1.