atc1441/ATC_MiThermometer #275: Similar sensor but with product code NUN4126GL

sanyi, 2022-12-28:
The previous LYWSD03MMC seems to be no longer available. However there is a slightly more expensive version NUN4126GL available. Exterior looks exactly the same (from the product pages) however I am not sure about the CPU and compatibility with prev versions. Are there anyone having this variant and/or any experience with it?

fastriker, 2023-01-24:
I would be interested as well.

ajbanck, 2023-02-15:
I just got the NUN4126GL, production date 11/2022.
The SKU is NUN4126GL, however according to the manual the model ID is LYWSD03MMC which is how it is published on Bluetooth.
Difference I see is that no battery is included. Also the packaging is different, manual is in six languages and statement the device conforms to EU standards.

fastriker, 2023-02-22:
That great news - thank you! Could you also flash a alternative firmware?

ajbanck, 2023-02-22:
> That great news - thank you! Could you also flash a alternative firmware?

Yes, no problems flashing with custom firmware. It's the same model, LYWSD03MMC