atc1441/ATC_MiThermometer #280: My thermometer does not connect via BLE

speerwerfer, 2023-03-12:
My thermometer does not connect via BLE. The connection keeps breaking. I have tried different browsers and devices. Even with new batteries it does not work.
Are there any instructions on how to flash the device via cable/UART?

dgoadby, 2023-03-13:
Are you using Windows? Are you using a laptop with internal Bluetooth? I got very frustrated with this myself. 

The solution for me was a new USB TP-Link Bluetooth dongle (UB-500 not 400) on a Linux laptop with the Chrome browser. 

The TP-Link USB dongle was good for Bluetooth 5 and worked like a charm. The laptop internal Bluetooth did not link as it was not to the latest spec for BLE.

I have now updated ten units with no problems so it was not a fluke!

I hope this works for you as the cable method is not the easiest thing to do.

speerwerfer, 2023-03-14:
I have already updated 4 thermometers with my setup. Also the Themometer, which now no longer connects stable, I have this already flashed with a custom firmware. So it is not because of my hardware.
I also do not want to experiment for a long time. I would now simply solder 4 lines and flash via cable.

dgoadby, 2023-03-14:
It does sound like a strange problem. It has not something that I have encountered. The only other item that I would check is the battery.