atc1441/ATC_MiThermometer #281: Latest LYWSD03MMC shows 129.9C  99% humidty after flashing

visago, 2023-03-21:
Recently got some new LYWSD03MMC and after flashing,  It shows the last 3 bytes of the mac address after booting, then it shows 129.9C / 99% humdity. I can still connect to it and set the face to be smiling,etc..

The broadcoast is showing similar values "Temp:129.9 Humidity:99.0 Batt:87.0" from my app.

I tried to flash back to original firmware from https://github.com/pvvx/ATC_MiThermometer/files/7300157/LYWSD03MMC_0130_upd_miaomiaoce.sensor_ht.t2.zip  and it displays the temp/humidity fine

Anyway to determine whats the hardware version ?

palmarci, 2023-12-29:
same problem here

tremichl, 2024-01-31:
almost the same here. Temperature is OK, but if "Show battery in LCD" is enabled, Humidity altering from 99% to correct value to 99% again and battery symbol is disappearing in same rhythm.

Anyway, thanks for great job

creamailer-ape, 2024-02-09:
> almost the same here. Temperature is OK, but if "Show battery in LCD" is enabled, Humidity altering from 99% to correct value to 99% again and battery symbol is disappearing in same rhythm.
> 
> Anyway, thanks for great job

I just flashed my first sensor and noticed same here. Did you find any solution for the altering humidity?

atc1441, 2024-02-09:
The 99 % is the battery status :)

atc1441, 2024-02-09:
This feature can be disabled in the Telin Flasher with the "Show battery in LCD" button

creamailer-ape, 2024-02-09:
> This feature can be disabled in the Telink Flasher with the "Show battery in LCD" button

Awesome! 🔥 Thank you for your work.