atc1441/ATC_MiThermometer #290: Feature request: remote display

mo8Zomo0, 2023-06-16:
It would be cool if 

* two sensors can be paired with each other to show the remote value.
like place one display outside (showing its own temperature) but also have an other one inside to show the outside temperature

and / or 

* use it as a remote display 
being able to remote control what is shown on the display. 
like being able to use some script to show whatever should be shown on the display, 
one usecase could be to use some home automation software like domoticz, home assistant, ... 
and show some value there, if its the outside or whatever measured temperature or other numerical / hex value.

njbuch, 2024-01-17:
Its now possible to show data from another sensor using homeassistant. Which mean you can show data from the other sensor on a specific one.

njbuch, 2024-01-17:
The pairing idea should go into a separate issue.

njbuch, 2024-01-17:
And this one can be closed.