atc1441/ATC_MiThermometer #296: LYWSD03MMC - can't connect after enabling "LE Long Range" even with BT5 USB adapter

tomwaldnz, 2023-09-09:
I use a half dozen Xiaomi LYWSD03MMC temperature sensors with Home Assistant, they work well. I have an [XDO bluetooth 5.0 adapter](https://www.amazon.com/Bluetooth-XDO-Wireless-Transmitter-Headphones/dp/B09KGVFX1K?th=1&tag=headphonerevi-20&linkCode=ur2&camp=1789&creative=9325), which I usually have connected to a Raspberry Pi 4, but also works on my Ubuntu PC.

As an experiment to see if I could get longer range I disabled "BT5+ PHY" and enabled "LE Long Range" on one of my thermometers. I can't connect to that thermometer any more, either with the XDO BT5 adapter, with a modern Samsung phone that also supports BT5, or with a Google Pixel 4a that also supports Bluetooth 5 LE.

The sensors are cheap enough that they can be replaced easily, but I'd like to see if I can get it working again. I'm not interested in cutting up a USB cable and soldering to get a physical serial interface though. Can anyone help? Or have I ticked options in a way that mean it's basically bricked?

Here's the details of my Bluetooth adapter

```
hciconfig -a
hci0:	Type: Primary  Bus: USB
	BD Address: 00:E0:42:8B:xx:xx  ACL MTU: 1021:6  SCO MTU: 255:12
	UP RUNNING PSCAN ISCAN 
	RX bytes:5737 acl:0 sco:0 events:341 errors:0
	TX bytes:32992 acl:0 sco:0 commands:303 errors:0
	Features: 0xff 0xff 0xff 0xfe 0xdb 0xfd 0x7b 0x87
	Packet type: DM1 DM3 DM5 DH1 DH3 DH5 HV1 HV2 HV3 
	Link policy: RSWITCH HOLD SNIFF PARK 
	Link mode: PERIPHERAL ACCEPT 
	Name: 'RTK_BT_5.0'
	Class: 0x000000
	Service Classes: Unspecified
	Device Class: Miscellaneous, 
	HCI Version: 5.1 (0xa)  Revision: 0xdfc6
	LMP Version: 5.1 (0xa)  Subversion: 0xd922
	Manufacturer: Realtek Semiconductor Corporation (93)
```

d1scolor, 2024-06-03:
Got the same problem

d1scolor, 2024-06-03:
OK fixed it by shorting the Reset and Gnd on the pcb (you can see them after you open the backplate sensor, next to the battery). It will uncheck that LE Long Range option. When you successfully short the two (when sensor is powered on), you'll see the screen briefly goes blank and it will restart automatically. After the reset, some of (maybe all of?) the other settings remain unchanged.

tomwaldnz, 2024-06-03:
Interesting. I solved the problem by throwing it in the rubbish bin and buying another, but your way seems better :)