atc1441/ATC_MiThermometer #298: Xiaomi has released new 2.1.1 firmware

micturkey, 2023-10-01:
[image]
It seems that the way to update firmware using telink flasher has been forbidden.

atc1441, 2023-10-01:
Thank you for the hint.
That really reads like they closed it

pvvx, 2023-10-10:
Not closed, but changed the "activation".
"Login", works with known keys. OTA also works.

A piece of the activation log:
"Sent Read Request, Handle: 0x0061" abbreviated as "Send 6C".
"Rcvd Read Response, Handle: 0x0061" abbreviated as "Rcvd 6C".
```
// > Checking for transmission with MTU size
Send 63: a4
Rcvd 6C: 000004000612
Send 6C: 000005000612
Rcvd 6C: 0000040112121212121212121212121212121212
Send 6C: 0000050112121212121212121212121212121212
Send 64: ? 
Rcvd 64: ? 
// >
Send 63: a2000000
Rcvd 6C: 000000000200
Send 6C: 00000101
Rcvd 6C: 01000200000000626c742e362e316667336a736f
Rcvd 6C: 0200726f73673030
Send 6C: 00000100
Send 63: 15000000
Send 63: 000000030400
Rcvd 6C: 00000101
Send 6C: 01003b412a2b060a1d7da21033ff4e584bf4f8f3
Send 6C: 02001a9dee5c4dc95e198c4bc3be5953d6babfdb
Send 6C: 0300415a9eda4e42ac53e864d1ebd6c9b4616ce5
Send 6C: 04004c9f1e094e30fc77ce51
Rcvd 6C: 00000100
Rcvd 6C: 000000030400
Send 6C: 00000101
Rcvd 6C: 010025c4faa9e119108b3133915e663ee3d4d0fb
Rcvd 6C: 02009ada216d9d91928725dea0bb88f44639f8a1
Rcvd 6C: 0300bb69a33f849bdbb0c2be2b8910271244c5dd
Rcvd 6C: 04006bc5edefc593dc2d8557
//> Updated part
Send 6C: 00000100
Send 63: 13000000
Send 6C: 000000000600
Rcvd 6C: 00000101
Send 6C: 01001904e3c44b8ab77b3e2f9b7371b4606d9a8a
Send 6C: 0200e7c71cc8bc712b7d080af2153d8638b7701e
Send 6C: 0300ab70d36fceb296c3f8805d4073216e542f93
Send 6C: 0400523da93c45061966487db32dd32936159b3e
Send 6C: 05006739aa0281d368eac3205bc87d419ebc838e
Send 6C: 06007457
Rcvd 6C: 00000100
Send 6C: 000000071600
Rcvd 6C: 00000101
Send 6C: 0100308201773082011ea003020102020101300a
Send 6C: 020006082a8648ce3d0403023022311330110603
Send 6C: 030055040a130a4d696a696120526f6f74310b30
Send 6C: 04000906035504061302434e3020170d31363131
Send 6C: 050032333038323032355a180f32303636313131 
Send 6C: 0600313038323032355a30233114301206035504
Send 6C: 07000a0c0b4d696a696120436c6f7564310b3009
Send 6C: 080006035504061302434e3059301306072a8648
Send 6C: 0900ce3d020106082a8648ce3d03010703420004
Send 6C: 0a00a752ecd44b6b3b17abc34f8300c6320f2e4c
Send 6C: 0b00bec57a51034b5ecadf7347d745df8c3dbcfa
Send 6C: 0c00aedb67b04cace5aff798182e43c5a444b627
Send 6C: 0d00c2d7f361629d3f914802a3423040301f0603
Send 6C: 0e00551d2304183016801496b7a27c39b1b96633
Send 6C: 0f00a9f8d109b20060c8e6c511301d0603551d0e
Send 6C: 1000041604145a29bffb2fb7500ce9c420f23d89
Send 6C: 11009b6fe0803293300a06082a8648ce3d040302
Send 6C: 1200034700304402205eb096d630f92f092ae39d
Send 6C: 13001356f836c529697a355d765f4eccce785b89
Send 6C: 14009a6d1602207e206b22aa04e6dee818c7d4c4
Send 6C: 150080e5fabd99074bdecf45346e37f1cffd8646
Send 6C: 160090
Rcvd 6C: 00000100
```

atc1441, 2023-10-10:
Thanks Victor, 
So the activation part is not cracked right now, and you need to currently get the set key first from the app etc. to OTA ? 

I would expect them to sign the activation on the server side with an unknown private key but lets hope not

thazro, 2023-10-10:
So for 2.1.1 no way to get keys for now, only if you previously had them. Can i downgrade via Uart?

pvvx, 2023-10-11:
I haven't clarified the whole process yet.
There is an assumption that the OTA firmware is signed with an additional key.
The "OTA" procedure itself always works, but at the end the "OTA" code itself may not be included. Previously, for some variants of thermometers, the signature was the correct "CRC" of the OTA code.

maltiboi, 2023-10-17:
is this going to get fixed please? thank you

Tim-The-Woodsman, 2023-10-19:
And I just updated the firmware without checking in here 😞

wwnkrull, 2023-10-20:
I've made the same mistake by updating to the latest firmware. Hope this will be fixed soon. 
Thanks guys for all the hard work!

PerssonNiklas, 2023-10-21:
So for those of us who updated to the latest firmware, is there any way to downgrade when the flasher does not connect due to being on unsupported firmware? Catch 22 situation!

igorlesiv, 2023-10-21:
I also can't  flat on 2.1.1_0159 version, let me know please it if possible or not. Thanks!

pvvx, 2023-10-21:
At the moment, you can only write another firmware using a hardware programmer.

pvvx, 2023-10-23:
A hint may occur when a new version is released. When it will be possible to upgrade version 2.1.1_0159 in Mi Home to the next one.

VonalOrdu, 2023-11-02:
To go back to the old version (Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin );
To do this, you must remove your temperature sensor and connect it with the cables by following the steps below.
This is how I solved my problem.
I'm sorry for my bad english.

1. https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin download
2. https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html 
3. [image]
4. [image]
5. File select Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin
6. Write to flash

pvvx, 2023-11-02:
@VonalOrdu The contact on the PCB marked as "reset" is not an RST signal for the TLSR825x SoC !

VonalOrdu, 2023-11-02:
> @VonalOrdu
> 
> 1. The contact on the PCB marked as "reset" is not an RST signal for the TLSR825x SoC !
> 2. https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html - does not use RX pin,  resistor and connection RX is not needed !

I have no idea about this.
I succeeded by doing this.
Maybe it will happen if you do as you say.
I used the suggestion below.

https://github.com/atc1441/ATC_MiThermometer/blob/master/Mi_SWS_Connection.jpg
[image]

pvvx, 2023-11-02:
The very name of the utility describes: _TLSR825x USB-COM Flash Writer v0.4_ (**TX-SWS only!**)  :)

VonalOrdu, 2023-11-02:
> The very name of the utility describes: _TLSR825x USB-COM Flash Writer v0.4_ (**TX-SWS only!**) :)
> 
> The picture is from another version of the programmer - https://github.com/pvvx/TlsrComProg825x - does not work on FTDI chips!

[image]

Are you saying that I am enough like this?

pvvx, 2023-11-02:
> Are you saying that I am enough like this?

Yes

VonalOrdu, 2023-11-02:
> > Are you saying that I am enough like this?
> 
> Yes

`python.exe TLSR825xComFlasher.py -p COM3 -t 70 wf 0 Original_full_flash_Xiaomi_LYWSD03MMC.bin `

Why didn't this method work?
"Chip sleep? -> Use reset chip (RTS-RST): see option --tact"
It was giving error.

https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html

This method worked. Thank you very much for your sharing.

pvvx, 2023-11-02:
> Why didn't this method work?

https://github.com/pvvx/TlsrComProg825x - **does not work on FTDI chips!**

---

https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version

[image]

ceinmart, 2023-11-07:
Hi, 
I just made the same error as everyone here.
I bought two of these sensors and was planning to flash them and use them with a BLE Tracker on my Home Assistant. 
However, curious to see how it works originally I did the firmware upgrade when added to mihome app.... stupid curious... 

So, after sharing my disgrace... 

How hard is to get one of this USB-COM board and use it to downgrade? any link from Aliexpress?
I get little confused about which board is compatible, which link should be used to do the downgrade, how is the right way to wiring... 
Please, can anyone share where to buy this USB-COM and a step-by-step how to downgrade the firmware ?

thazro, 2023-11-08:
> Hi, I just made the same mistake as everyone here. I bought two of these sensors and was planning to flash them and use them with a BLE Tracker on my Home Assistant. However, curious to see how it works originally I did the firmware upgrade when added to mihome app.... stupid curious...
> 
> So, after sharing my disgrace...
> 
> How hard is to get one of this USB-COM board and use it to downgrade? any link from Aliexpress? I get little confused about which board is compatible, which link should be used to do the downgrade, how is the right way to wiring... Please, can anyone share where to buy this USB-COM and a step-by-step how to downgrade the firmware ?

Hi. Downgraded using this ch340 usb to ttl rs232 converter:
 https://www.aliexpress.com/item/32354359382.html?gatewayAdapt=glo2isr
Solder p14 on thermometer to txd
Solder Gnd to gnd
Solder + to 3.3V
Flash using:
https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html

kimol88, 2023-11-09:
Maybe I can't differently, but flash by site only works on Windows "machine". On MacBook I bricked by flash. On Windows "machine" I recovery firmware without problems :)

ceinmart, 2023-11-12:
> Hi. Downgraded using this ch340 usb to ttl rs232 converter: https://www.aliexpress.com/item/32354359382.html?gatewayAdapt=glo2isr Solder p14 on thermometer to txd Solder Gnd to gnd Solder + to 3.3V Flash using: https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html

Hi Guys, 
Just giving an feedback, I just flashed and downgraded my sensor successfully!!!

However I didn't used the board referenced by @tharzo, I got with a  Friend the FTDI 232 and followed the steps passed by tarzho.

I did the downgrade of my firmware from v2 to v1 and then flashed it with the custom firmware v4.5 successfully!! 
Very , very happy :)
Thanks all for the support.
Now , let's try to setup it on my Home Assistant using a ESP32 as BLE Tracker...

[image]

[image]

[image]

[image]

vdende, 2023-11-13:
I'm not that familiar with soldering and boards, so I decided to buy a new one from Ali, from the same shop as my previous one. It was shipped very fast and fortunately the firmware version of the new device was still on v1.0. 
So for this one I was able to flash it with the custom v4.5, set it to BTHome and configure it in HomeAssistant.

adamb94, 2023-11-25:
Is there any expected date when soft 2.1.1 will be supported by Telink Mi Flasher? I was not able to downgrade by Serial

pvvx, 2023-11-25:
So far no one is doing this or it is unknown.
I'm waiting for the next version to come out.
This will make it possible to understand how to update version 2.1.1.

atc1441, 2023-11-25:
While i dont know 100% i am veeery sure they added a signing to the update which can not be bypassed without any exploit 

So ota is unlikely.

Just my assumption

pvvx, 2023-11-25:
> While i dont know 100% i am veeery sure they added a signing to the update which can not be bypassed without any exploit

They could also add key reading to their cloud for a registered user.

atc1441, 2023-11-25:
> Disassembling or otherwise viewing codes from Xiaomi is prohibited in the MiHome user agreement. For this reason, other methods that are not prohibited will be used. And this requires the next new version of OTA from MiHome.

I did never agree to anything like that so no problem with disassambly ^^

pvvx, 2023-11-25:
Using information from open sources is not prohibited...  -> Need a publication :)

rmappleby, 2023-11-26:
I have also never installed MiHome (or anything else from Xiaomi), and so am not subject to any of their terms and conditions. I assume @atc1441 will provide the information that is needed far more effectively than I can, but if not give me a nudge.

rezmus, 2023-11-26:
> And this requires the next new version of OTA from MiHome.

i can provide. any way to pm you on telegram or something?

atc1441, 2023-11-27:
Yep thats a singed OTA :(

rmappleby, 2023-11-27:
But there's still nothing to stop us flashing your custom firmware using wires and an FTDI converter, I assume?

atc1441, 2023-11-27:
@rmappleby yes, the SoC can always be flashed via wires

pvvx, 2023-11-27:
[Original_OTA_SJWS01LM_1.1.1_0018.bin](https://github.com/pvvx/ATC_MiThermometer/blob/master/Original_OTA_SJWS01LM_1.1.1_0018.bin)
-----BEGIN CERTIFICATE-----
MIIBhTCCASugAwIBAgIBAjAKBggqhkjOPQQDAjAiMRMwEQYDVQQKEwpNaWppYSBS
b290MQswCQYDVQQGEwJDTjAgFw0xODAxMTgwMjE5MTVaGA8yMDY4MDEwNjAyMTkx
NVowIjETMBEGA1UECgwKTWlqaWEgT3BlbjELMAkGA1UEBhMCQ04wWTATBgcqhkjO
PQIBBggqhkjOPQMBBwNCAASTPDXoepwRzEV0nVioPiILRzjKTNZo1dUSzutyn8op
V4NaQDmCwtdvGGz/UzWEqEdlgVuncnG8S8de1mVxZhzIo1AwTjAfBgNVHSMEGDAW
gBSWt6J8ObG5ZjOp+NEJsgBgyObFETAdBgNVHQ4EFgQU6ik3eeFQQohlpo7V0Ehr
wLBT+GUwDAYDVR0TBAUwAwEB/zAKBggqhkjOPQQDAgNIADBFAiEAnQyFwFhYV/Zm
n5OY0H4FG7qVarqVUMrXiNCg/cIif+ACIEW1J06zPTYNLe9SqIIaFiCBMTtvPUvM
JJHkc890sLAY
-----END CERTIFICATE-----
-----BEGIN CERTIFICATE-----
MIIBlzCCAT2gAwIBAgIGAXRoWA0NMAoGCCqGSM49BAMCMCIxEzARBgNVBAoMCk1p
amlhIE9wZW4xCzAJBgNVBAYTAkNOMCAXDTIwMDkwNzExMzM1NVoYDzIxMjAwODE0
MTEzMzU1WjAcMQswCQYDVQQGEwJDTjENMAsGA1UEChMEY2VydDBZMBMGByqGSM49
AgEGCCqGSM49AwEHA0IABF1W2an4g7STG3WAPEo5GyADHqIG+78VehOcVKnj/My3
krQf6jJ6tpE9krOvI5djAi18rGbfLK5xmLRB3p6ybM6jYzBhMA4GA1UdDwEB/wQE
AwIFoDATBgNVHSUEDDAKBggrBgEFBQcDATAMBgNVHRMBAf8EAjAAMB8GA1UdIwQY
MBaAFOopN3nhUEKIZaaO1dBIa8CwU/hlMAsGA1UdEQQEMAKCADAKBggqhkjOPQQD
AgNIADBFAiAY0qwBcD9Tse8VayhUjFyMFeNOD3/4pozRP7WygCJYRgIhAMac6yq2
vh54uDyirfDu3YvLCyISIvkO1wW0ytsj2Xkp
-----END CERTIFICATE-----

volodymyr-koval-vitech, 2023-11-30:
Using instruction from @thazro with CH340C I've successfully downgraded firmware, but only under Windows as @kimol88 mentioned. Take this into consideration. Who knows what can be issue under Mac?

pvvx, 2023-12-01:
> Who knows what can be issue under Mac?

None of the developers of alternative firmware work on Mac.

volodymyr-koval-vitech, 2023-12-07:
I had similar symptoms when tried to flash on Mac. Which OS are you using to flash?

besanur, 2023-12-07:
> I had similar symptoms when tried to flash on Mac. Which OS are you using to flash?

I am using macos,i have tried it on a window vm with no success. I'll try it with a windows computer.

volodymyr-koval-vitech, 2023-12-07:
MacOS doesn't work. I've successfully flashed from Windows run under Parallels Desktop.

besanur, 2023-12-07:
> MacOS doesn't work. I've successfully flashed from Windows run under Parallels Desktop.

Can you pls tell me how you wired it. Like me ? 
TX -> P14
GND -> GND
3V3 -> V+

TX without resistor ?

pvvx, 2023-12-07:
The USBCOMFlashTx.html program does not require any resistors.
https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version

pvvx, 2023-12-07:
> Can you pls tell me how you wired it. Like me ?

https://github.com/pvvx/ATC_MiThermometer#the-usb-com-adapter-writes-the-firmware-in-explorer-web-version

[USBCOMFlashTx.html](https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html)
The USB-COM adapter writes the firmware in explorer. Web version.
Connect only TX-SWS and GND wires.
[image]

In difficult cases, when the chip contains an unknown FW, it is necessary to use a chip reset or power supply during "activation".

[image]

At the same time, take into account that the chip can be powered from the TX output. 
**This is solved by connecting TX and +Vbat during "activation".**

pvvx, 2023-12-07:
Requires chip restart/reset during "activation".
For any programmer.
How you provide it is your task. 
By turning off the power, turning off all extraneous sources, or by shorting the RST pin of the TLSR825x chip itself to GND.

jjvelar, 2023-12-09:
Hi!
Any news on LYWSD03MMC with firmware version: 2.1.1_0159?
I just can't connect. Pop up message says: "couldn't find any compatible device".
Thanks in advance for your help.

pvvx, 2023-12-10:
> Hi! Any news on LYWSD03MMC with firmware version: 2.1.1_0159? 

We are waiting for the popularity of Xiaomi LYWSD03MMC to fall, sales and prices to fall. :)

jjvelar, 2023-12-10:
Thanks for replying @pvvx 
Any good tutorials on how to roll back to a previous firmware version?
I'm pretty much a newbie.
Thanks in advance.

Rawi666, 2023-12-14:
Thank you all very much! I successfully downgraded the firmware without soldering! I ordered a cheap USB-TTL converter.
I used left hand to hold steadily VCC and TX pin (near P14/P10 hole on the PCB), it took me a minute practice to stabilise it. Then I set the flash timer to have 8 seconds delay, I clicked flash and then within 8 seconds I grabbed my right hand and holded GND pin :-D
After 42 seconds holding hands steadily I was able to finish flashing successfully. No soldering needed guys!
I think even if you screw the flashing you can start over and reflash from a broken software, but someone would need to confirm it :)

pvvx, 2023-12-14:
> I think even if you screw the flashing you can start over and reflash from a broken software, but someone would need to confirm it

Reprogramming can be launched a thousand times. This will not even reduce the Flash resource, since the manufacturer guarantees more than 100 thousand rewritings.
[image]

My trials of various programs on one of the Xiaomi Lywsd03mm have long exceeded a couple of thousand flash overwriters. It works and is used for debugging. A normal programmer with a hardware Telink SWire writes a firmware for a couple of seconds.

ffries, 2023-12-15:
After flashing with OTA the display does not show and the thermometer does not announce itself even after shoring. Does it mean that it is bricked?

pvvx, 2023-12-15:
> After flashing The Xiaomi with OTA, the display does not show and the thermometer does not announce itself even after shorting pins. Does it mean that the XIaomu is bricked?

Firmware for OTA version 2.1.1?

Symbiot78, 2023-12-30:
chiming in with the same 2.x firmware issue. Hoping everything "we" are waiting for will come out soon :-) Keep up the great work!

Hubszo, 2024-01-02:
Hi,
I have Qingping CGDK2 (round one), which was previously successfully flashed with custom firmware 4.5. Yesterday I OTA flashed it with Zigbee Firmware: ZCGDK2_v0119.bin with Telink Flasher to convert from BLE to Zigbee and everything went ok according to the log but now the sensor is stuck on -39°C and 100% humidity. Reseting by button or battery removal didn't help. Any ideas???

pvvx, 2024-01-02:
>  Any ideas?

Flash back to BLE using Zigbee OTA.

> ZCGDK2_v0119.bin

This doesn't exist yet. There is ZCGDK2_v0118.bin

Hubszo, 2024-01-02:
That was my typo. I meant 0118.bin, of course.

pvvx, 2024-01-02:
About Zigbee OTA, see https://github.com/pvvx/ZigbeeTLc

apeelme, 2024-01-05:
> What did I do wrong when flashing? Nothing is shown on the display after I used https://pvvx.github.io/ATC_MiThermometer/USBCOMFlashTx.html to flash.
> 
> My connection from the [UART-TTL USB Adapter](https://www.az-delivery.de/products/usb-ttl-adapter-3-3v-oder-5v-mit-ch340g-und-20cm-jumperkabel) to the thermostat was like this. TX -> P14 GND -> GND 3V3 -> V+
> 
> I tried several times to flash the original firmware but also the zigbee version. Erase all Flash didn't help either

Same happend with me. :(

gjesch, 2024-01-07:
I have two issues with the Xiaomi LYWSD03MM 
_(I ordered 4 of them plus 1 CGDK2. Unfortunately I upgraded 2 of them to 2.1.1, well ...)_

(A)
I flashed with the UART method the ZigBee firmware 1.0.6 from https://github.com/devbis/z03mmc/releases to 1 device (1 out of the 4 devices). The device hardware works OK, but ZHA just updates the temperature. The humidity stays fix on 50%. I will report this on devbis/z03mmc/discussions as well: https://github.com/devbis/z03mmc/discussions/104#discussion-6041570. Just in case anyone has already observed such a behaviour.

(B)
I want to stay on BLE for the other 3 devices in order to compare battery life and overall experience. Therefor I have to downgrade to **Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin** again, or flash **ATC_v38.bin** right away. 

But although I see these firmware versions being used in various screenshots - also on this page - I get the message "**Select file is no telink firmware .bin**"

[image]

I must obviously be missing something, but what?
Thanks for any advice.

[300 more replies]