atc1441/ATC_MiThermometer #299: Type Error: GetDevVersion()

Trikos, 2023-11-04:
Hi,

I flash my T&H mi one years ago using your software, today I want to try to update the firmware in order to remove the battery icon and enable BTHome to connect on Home assistant but when I try to connect to it I get this error on console.
What should I do?
`00:37:28: Searching for devices
00:37:44: Connecting to: ATC_XXXXXX
00:37:48: TypeError: getDevVersion(...).then is not a function
00:37:48: Reconnect 1 from 5
00:37:48: TypeError: getDevVersion(...).then is not a function
00:37:48: Reconnect 2 from 5
00:37:48: TypeError: getDevVersion(...).then is not a function
00:37:48: Reconnect 3 from 5
00:37:48: TypeError: getDevVersion(...).then is not a function
00:37:48: Reconnect 4 from 5
00:37:48: TypeError: getDevVersion(...).then is not a function
00:37:48: Reconnect 5 from 5
00:37:48: TypeError: getDevVersion(...).then is not a function
00:37:48: Something went wrong, too many reconnects
00:38:05: Disconnected.`

Thank you

sanderlv, 2023-11-11:
I have EXACTLY the same situation here... I tried to flash manually with bin rlease 70 and relase 78 but everytime same problem...

romka777, 2023-11-11:
I had the same problem, i think it was because flashed firmware with hw flow control. 
Fixed by this video https://www.youtube.com/watch?v=CympfXuwtyQ