atc1441/ATC_MiThermometer #310: Compatibility with TH05?

RChadwick7, 2023-12-30:
I purchased a number of TH05 devices on Aliexpress. Using the OTA flasher, I see them, but can't pair with them. Is there a way to flash these devices OTA? Or is a UART the only way?

Also, if it's the only way, will they be OTA flashable after having open source firmware flashed to them?

Thanks!

almirus, 2024-01-22:
@RChadwick7 not yet https://pvvx.github.io/TH-05/
but work in progress. there is beta BIN for TH-02 https://pvvx.github.io/THB2/

renedis, 2024-04-03:
https://pvvx.github.io/TH-05/
https://github.com/pvvx/THB2/tree/master/bin
https://github.com/pvvx/THB2/blob/master/bin/TH05_v16.bin

There is already a version 1.6... But I could not get it to work.

madhatter349, 2024-10-21:
Big fan of the Telink Flasher for Mi Thermostats—I've successfully flashed several using this method. I recently picked up a few Tuya TH05 units and was wondering if there have been any updates on flashing these OTA?

man1k, 2025-01-16:
Hi there,

Is there any chance of getting this working for the TH05?