atc1441/ATC_MiThermometer #317: ad interval setting seems to not work

mickio, 2024-02-05:
The telink flasher site allows to set the advertising interval for flashed devices. While it is possible to toggle battery consumption display, the alteration of the advertising interval does not show any effect, though the log tells e.g. "Settings FE06 was send successful". For longer battery life I would like to set the interval to 10min.

thomasabbott, 2024-02-08:
I have this too. 5-second packets no matter what interval I ask for.
But see comments on #302 - is something like your ESP polling the devices actively?

thomasabbott, 2024-02-12:
I find the LYWSD03MMC transmits every 1.8 - 3 seconds. Two or three packets.

Configuration:
- Flashed easily with the web flasher
- Firmware: Release 79 of Dec 24, 2023
- HW version... I don't have a small enough Torx, these were bought in 2024.
- Temperature offset configured, FAFB
- Degrees C
- Report on change FC and FD is set to a large number, FC66 and FD66
- Smiley off

Things I've tried changing:
- Advertising type Mi or Custom, no change
- When set to 1 minute or 5 minute reporting interval (always saved in flash)
- No influence from nearby devices - the theory that something was making a connection to it.  
I tried with and without the nearby ESP32+Tasmota, no change
I placed the MiTherm and an RF amplifier + detector + LED in a microwave oven, power cycled and quickly closed the door. Same behaviour, transmitting every ~2 seconds. My phone could detect the device from nearby but not from a few metres away, so it's unlikely anything was interrogating it in there.
- Temperature updates on the display every 5 seconds or so

My device reported battery is plummeting, 70% after 3 weeks, but not yet run out so I can't say for sure.

My goal is to increase battery life, I don't care about the packets otherwise.
Does it help to reduce the reporting (and measuring) rate?
Is there a way to do this, have I done something wrong?
Thanks!