atc1441/ATC_MiThermometer #320: Force into pairing mode.

gheydon, 2024-03-24:
Hi,

I had a problem with my Zigbee network, and had to rebuild it using a new radio. Most devices have a pairing button or in the case of some lights you turn them off and on x number of times and they enter pairing mode.

I have a couple of these sensors and they worked great, but now that my network has been rebuild I need to re-pair them, to the Zigbee network and I am unable to find how to get them into paring mode.

I have tried to re image the device but I can't connect it it anymore via bluetooth and it's not on the Zigbee network so I can't upgrade via that.

Any instructions on how to get it into pairing mode will be very much appreciated.
Gordon.

ur3an0, 2024-06-30:
UP

atc1441, 2024-06-30:
This repo is for the BLE Firmware only.
Please look out from where you got the zigbee firmware

closed 2024-06-30 by atc1441