atc1441/ATC_MiThermometer #325: Device bricked after successful flash and battery removal

82MHz, 2024-05-23:
Hi, bit of a long shot here... I flashed the device OTA with the latest release version, and it booted up fine afterwards, showed the last bytes of the MAC address as expected, everything seemed good. I noticed the battery indicator showed low battery, so I took the battery out, put a new one in and now the device seems dead, there is no activity on the screen anymore. 

Any idea how I could revive this?

siradlib, 2024-06-29:
Hi, which ATC firmware version did you use?  I made the same experience here with v48 beta....
...but after disconnecting for 5 mins and connectíng the battery again, my device Xiaomi lywsd03mmc came back to life.
Give it a try!

aptalca, 2024-08-12:
On the latest firmware, after a battery death and replacement, the screen never comes back on for me. But I can still connect over bt.
Reflashing the latest doesn't make a difference.

The only thing that fixes it is to first flash the original manufacturer firmware, and then flashing the latest firmware from here.