atc1441/ATC_MiThermometer #329: Flashing not working.

vidalv, 2024-06-25:
Hello, I try to flash my part without success. I use first the chrome interface on my laptop, after connection it's impossible to start flashing : 
[image]
Looking on issues on the git I found another utility : 
[image]
With this it seems the firmware version 2.1.1_0159 is not supported, any other solution please ? 
Thanks.

vidalv, 2024-06-25:
Thanks @atc1441, argh!!! too bad ! No work around .... Any link to do that ?

atc1441, 2024-06-25:
No work around right now: https://github.com/atc1441/ATC_MiThermometer?tab=readme-ov-file#usb-to-uart

vidalv, 2024-06-25:
Ok thanks, after hard flash is it possible to flash (for future need) with OTA ?

atc1441, 2024-06-25:
Yes

vidalv, 2024-06-25:
Super ! Thanks again.

vidalv, 2024-06-25:
Just a comment about wiring, I'm Ok for VCC (3.3V) and GND, but snapshot is disappointing : 
[image]
Only One wire for TXD and RXD (no resistor ?!!?) Could you clarify ? 
I would like to use my USB to TTL (UART) converter but USB is 5V not 3.3V ... This can damage the UC.

atc1441, 2024-06-25:
Yes you can leave out the resister and connect directly
Please use 3.3V Uart converter

vidalv, 2024-06-26:
I've found this : 
[image]
Is SWS same as P14 (RXD / TXD), I want to use this USB/UART : 
[image]
But is a FTDI module, is it usable ? Thanks.