atc1441/ATC_MiThermometer #332: Startup MAC ADDRESS missing

skydiverscott, 2024-08-01:
After manually and successfully updating the firmware via USB serial on the new HW B2 devices with ATC_4.7 I was able to change some settings like Fahrenheit, show battery, esp home advertismniug.  The one issue that may be in configuration that I cannot seem to find is that when the unit is restarted by replacing the battery, the start screen goes strait into the temp screen instead of splashing the last 6 of the MAC address.
Can this be turned on in the 4.7 config screen or is this not part of the new firmware?

skydiverscott, 2024-08-09:
Can anyone provide some insight as to whether this functionality is available or not in the 4.7 firmware?

rossi75, 2024-08-13:
I just installed the firmware 4.7 on three fresh B1.5 and I am facing the same issue