atc1441/ATC_MiThermometer #334: Humidity unstable (jumps 30, 60, 30, 60...) with this firmware

andrey-utkin, 2024-09-14:
I flashed my LYWSD03MMC with Release 79.

Temperature indication on screen is reasonable, but humidity on screen jumps around every 10 seconds: 30-something, 60-something, 30-something, 60-something, and so it goes.

This doesn't seem to change with settings available on your web flasher.

Flasing original firmware Original_OTA_Xiaomi_LYWSD03MMC_v1.0.0_0130.bin from pvvx restores the normal behaviour.

BTW @atc1441 could you please share the original firmware dumps you yourself dumped?

atc1441, 2024-09-14:
The jumping is the battery percentage, every time the battery icon is shown the humi nimber is the battery :)

andrey-utkin, 2024-09-14:
Ah! Thanks :)

closed 2024-09-14 by andrey-utkin