atc1441/ATC_MiThermometer #339: Provision Bind Key to other Xiaomi devices, e.g. Xiaomi Motion Sensor 2S (XMPIR02SGXS)

rmeissne, 2025-01-04:
Victor,

Thanks for your work on this. Can this be extended to provision the bind key to e.g. Xiaomi Motion Sensor 2S (XMPIR02SGXS). I understand that the TelinkMiFlasher does "fake the initial connection of the sensor with the Xiaomi cloud". Can this be done for other devices? How does this actually work? Is it possible to generate a bind key and "write it to the device".

I have the Xiaomi Motion Sensor 2S (XMPIR02SGXS), added it to the Xiaomi app, have received the bind key with the token extractor. I do not have the Xiaomi gateway. With the phone, I can change the settings on the device, sensitivity, timings. This works. The sensor however does not provide any BLE updates.

I strongly assume, that for this initial setup, you need to have connectivity to the cloud as this bind key is updates from the cloud - via the gateway to the device. I did assume, this would work via the phone app as the app could retrieve the bind code from the cloud and push it to the device.

Is this bind key written to some UUID via a BLE update? How does this actually work, e.g. for the LYWSD03MMC.

Thanks,
Robert

atc1441, 2025-01-04:
Hey Robert,

I am Aaron and not Victor but thank you anyway!

You can use this tool to activate other devices: https://atc1441.github.io/Temp_universal_mi_activate.html

But the compatibility is not tested so might not work on your device

rmeissne, 2025-01-04:
Hi Aaron, sorry for the mixup.

The tool does actually connect to the device (the other ones did not).

21:50:27: Searching for devices
21:50:34: Connecting to: xiaomi.motion.pir1g
21:50:36: Detected Mi Thermometer
21:50:36: Connected
21:50:43: Activating now, please wait...

It does show the device id "blt.4.1k1qppt8pgk01".

However, even after several minutes the token/bind-key fields are not populated. Can we somehow troubleshoot this please? Where does this hang? Is this tool randomly generating a bind key? This is an update to some UUID on the device?

Thanks a lot!
Robert

atc1441, 2025-01-04:
No worries.

As mentioned it is possible that the tool is not compatible, you will be on your own debugging it, if Xiaomi did change anything it will be a long procedure of debugging any finding out what is needed

rmeissne, 2025-01-04:
Understand. Any chance printing more debug messages to get more insides into "22:09:01: Activating now, please wait..."? What is the tool actually doing once the activation is initiated?

Thanks,
Robert

atc1441, 2025-01-04:
You can check out what is happening by looking at the source code of the .html page, its all open viewable :)

rmeissne, 2025-01-04:
Thanks. Yes, doing that right now ... I will try to figure out where it hangs :)

rmeissne, 2025-01-04:
Getting this far:

[image]

rmeissne, 2025-01-05:
I figured it out. It was actually my fault. I had a script that is matching on the device name. This worked for some beacons. But when the device does actually send "motion detected" and "lux values", the name is undefined. So my script did not present the messages.

This means - you can provision the sensor without the hub. It all works via the phone. The device does have the bind code programmed, I can read it out from the Xiaomi cloud.

The device does send v4 beacons. I wrote some script which does decode the values:

0201061a1695fe58585b05412fcfff38c1a45a3eeff41a0100009fe3bde7
String OK!
Length validated!
Valid HEX!
Hex Data:  020106161695fe4859604cf62c68d3237ce516020000a3d62777
Bind Key:  9d997e66dcb742b41059a22c3850a852
Nonce:  7194467744f8604cf6020000
Encrypted_payload:  2c68d3237ce516
MIC:  a3d62777
Cleartext:  084a040000ea42

084a040000ea42 - this is object ID 4a08 - "Motion detected with Illuminance in lux", 4 bytes: 0000ea42

Thanks,
Robert